SensorsIot/Definitive-Guide-to-IR #3: irsend keine reaction

malebuffy, 2019-04-05:
Funktioniert es noch? Habe ein ESP32 Dev Board und benutze für das IR Led, Pin 34 (andere auch versucht). Wenn ich irsend schicke, leuchtet das IR led nicht unter der Handykamera, also keine Reaktion.  Habe dein Code benuzt, ohne den MQTT teil, also was er als Input bekommt, schickt er wieder raus. Inpout funktioniert, output nicht. Hast du da eine Idee?

```
#include <IRremote.h>
#include <ArduinoJson.h>

int RECV_PIN = 22;
int SEND_PIN = 34;
int STATUS_PIN = 23;

decode_results results;

IRrecv irrecv(RECV_PIN);
IRsend irsend(SEND_PIN);

// Storage for the recorded code
int codeType = -1; // The type of code
unsigned long codeValue; // The code value if not raw
unsigned int rawCodes[RAWBUF]; // The durations if raw
int codeLen; // The length of the code
int toggle = 0; // The RC5/6 toggle state

void storeCode(decode_results *results) {
  // Stores the code for later playback
  // Most of this code is just logging
  char buf[10];
  String cType = "";
  String IRcommand = "";
  DynamicJsonBuffer jsonBuffer(200);
  JsonObject& root = jsonBuffer.createObject();

  codeType = results->decode_type;
  //int count = results->rawlen;
  if (codeType == UNKNOWN) {
    Serial.println("Received unknown code, saving as raw");
    codeLen = results->rawlen - 1;
    // To store raw codes:
    // Drop first value (gap)
    // Convert from ticks to microseconds
    // Tweak marks shorter, and spaces longer to cancel out IR receiver distortion
    for (int i = 1; i <= codeLen; i++) {
      if (i % 2) {
        // Mark
        rawCodes[i - 1] = results->rawbuf[i] * USECPERTICK - MARK_EXCESS;
        Serial.print(" m");
      }
      else {
        // Space
        rawCodes[i - 1] = results->rawbuf[i] * USECPERTICK + MARK_EXCESS;
        Serial.print(" s");
      }
      Serial.print(rawCodes[i - 1], DEC);
    }
    Serial.println("");
  }
  else {
    if (codeType == NEC) {
      Serial.print("Received NEC: ");
      if (results->value == REPEAT) {
        // Don't record a NEC repeat value as that's useless.
        Serial.println("repeat; ignoring.");
        return;
      } else cType = "NEC";
    }
    else if (codeType == SONY) {
      Serial.print("Received SONY: ");
      cType = "SON";
    }
    else if (codeType == PANASONIC) {
      Serial.print("Received PANASONIC: ");
      cType = "PAN";
    }
    else if (codeType == JVC) {
      Serial.print("Received JVC: ");
      cType = "JVC";
    }
    else if (codeType == RC5) {
      Serial.print("Received RC5: ");
      cType = "RC5";
    }
    else if (codeType == RC6) {
      Serial.print("Received RC6: ");
      cType = "RC6";
    }
    else {
      Serial.print("Unexpected codeType ");
      Serial.print(codeType, DEC);
      Serial.println("");
      cType = "UNKNOWN";
    }
    codeValue = results->value;
    codeLen = results->bits;
    Serial.println(results->value, HEX);
    Serial.println(results->bits);
    root["type"] = codeType;
    root["value"] = results->value;
    root["length"] = results->bits;
    root.printTo(IRcommand);
    root.prettyPrintTo(Serial);
  
  }
  
  sendCode(codeType, codeValue, codeLen);
  digitalWrite(STATUS_PIN,!digitalRead(STATUS_PIN));
  Serial.println("");
  Serial.println("Sending code");
  delay(500);
  digitalWrite(STATUS_PIN,!digitalRead(STATUS_PIN));
}

void sendCode( int codeType, unsigned long codeValue, int codeLen) {
  if (codeType == NEC) {
    irsend.sendNEC(codeValue, codeLen);
    Serial.print("Sent NEC ");
    Serial.println(codeValue, HEX);
    Serial.println(codeLen);
  }
  else if (codeType == SONY) {
    irsend.sendSony(codeValue, codeLen);
    Serial.print("Sent Sony ");
    Serial.println(codeValue, HEX);
  }
  else if (codeType == PANASONIC) {
    irsend.sendPanasonic(codeValue, codeLen);
    Serial.print("Sent Panasonic");
    Serial.println(codeValue, HEX);
  }
  else if (codeType == JVC) {
    irsend.sendJVC(codeValue, codeLen, false);
    Serial.print("Sent JVC");
    Serial.println(codeValue, HEX);
  }
  else if (codeType == RC5 || codeType == RC6) {
    toggle = 1 - toggle;
    // Put the toggle bit into the code to send
    codeValue = codeValue & ~(1 << (codeLen - 1));
    codeValue = codeValue | (toggle << (codeLen - 1));
    if (codeType == RC5) {
      Serial.print("Sent RC5 ");
      Serial.println(codeValue, HEX);
      irsend.sendRC5(codeValue, codeLen);
    }
    else {
      irsend.sendRC6(codeValue, codeLen);
      Serial.print("Sent RC6 ");
      Serial.println(codeValue, HEX);
    }
  }
  else if (codeType == UNKNOWN /* i.e. raw */) {
    // Assume 38 KHz
    irsend.sendRaw(rawCodes, codeLen, 38);
    Serial.println("Sent raw");
  }
}

void setup()
{
  Serial.begin(115200);
 
  irrecv.enableIRIn(); // Start the receiver

  StaticJsonBuffer<200> jsonBuffer;

  pinMode(STATUS_PIN, OUTPUT);
  digitalWrite(STATUS_PIN,LOW);
  Serial.println("Setup done");
}

void loop() {
  if (irrecv.decode(&results)) {

    storeCode(&results);
    irrecv.resume(); // resume receiver
   
  }
  
}
```

malebuffy, 2019-04-05:
Ich habe es mit dem Pin5 versucht, und es funtkioniert. Warum aber Pin 5? ist es ein ESP32 Hardware Thema oder funktioniert deine Version der IRremote Library nur mit Pin 5?

SensorsIot, 2019-04-10:
Ehrlich gesagt weiss ich es nicht mehr :-(

closed 2019-04-10 by SensorsIot