SensorsIot/Definitive-Guide-to-IR #7: IR receiver as Interrupt

Saivamsi075, 2019-10-12:
When we have big code and number of other tasks running in loop function the response to the IR signal gets slow or lost. So, I tried to put IR receiver pin as interrupt to esp8266 or to esp32 but, it does not worked. How can we give IR receiver pin as interrupt either to esp8266 or esp32?

Saivamsi075, 2019-10-12:
> When we have big code and number of other tasks running in loop function the response to the IR signal gets slow or lost. So, I tried to put IR receiver pin as interrupt to esp8266 or to esp32 but, it does not worked. How can we give IR receiver pin as interrupt either to esp8266 or esp32?

Saivamsi075, 2019-10-12:
Interrupt process is working with arduino uno but not working with esp8266 or esp32. Please help me

SensorsIot, 2019-10-12:
I do not know and unfortunately I do not have the time to do such a project. Maybe you find another library?

closed 2019-10-12 by SensorsIot