SensorsIot/Definitive-Guide-to-IR #11: Can not compile, I am begginer !!!

Formingus, 2023-10-16:
In file included from C:\Users\Formingus\Documents\Arduino\libraries\IRremoteESP8266\src/IRac.h:15:0,
                 from C:\Users\Formingus\AppData\Local\Temp\.arduinoIDE-unsaved2023916-5488-1m904gi.q7jqf\IRrecvDumpV2\IRrecvDumpV2.ino:33:
C:\Users\Formingus\Documents\Arduino\libraries\IRremoteESP8266\src/ir_Argo.h:14:10: fatal error: set: No such file or directory
 #include <set>
          ^~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

Formingus, 2023-10-16:
[https://youtu.be/gR3XOFoOLfA](url)  Hear is video when failed with last version 2.2.1

SensorsIot, 2023-10-16:
This is an old project, and maybe it does no more work.
If you want to try you have to copy the library folder of my Repo (ESP32-IRremote) to the library folder of Arduino. Then it compiles for the ESP32

closed 2023-10-16 by SensorsIot

Formingus, 2023-10-16:
> ESP32-IRremote

There is no library folder !! Examples, src, pictures are inside IRemote4.2.0.zip ? If you are talking about this file? If not where to download or advise me, on which project to use for Arduino Nano so i can read remote TV codes? Thanks in advance

Formingus, 2023-10-16:
[image]

SensorsIot, 2023-10-16:
Here is the structure inside the zip file: 
[image]

The ESP32-iRemote folder is a library

Formingus, 2023-10-16:
[image]

[image]

[image]

Which one to choose ?
[image]

SensorsIot, 2023-10-16:
As I wrote before: The ino file is in one folder and the library is in the other. Copy the library to Arduinos library folder and start the ino file. Then it should compile.

Formingus, 2023-10-16:
> As I wrote before: The ino file is in one folder and the library is in the other. Copy the library to Arduino's library folder and start the ino file. Then it should compile.

Here you dont have INO file !! ARE YOU TALKING ABOUT IRMQQTESP32 FOLDER?
[image]
 Ino Files are inside the example Folder. Can you tell me which one to open ???

[image]

If you want to help me please be more specific, i am a beginner --- You cant talk with me as advanced, i barely understand.

So, which one to open (NAME), and if this project doesn't work IS THERE SOME OTHER PROJECT THAT CAN RECEIVE REMOTE TV CODRS VIA ARDUINO NANO?? THANKS

SensorsIot, 2023-10-16:
Yes. You di not need another library. It is in my Github